bambulab/BambuStudio #2272: Stray tree supports

whatthefilament, 2023-08-11:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.0

### Additional system information

M1 Mac Mini

### Printer

X1C 

### How to reproduce

When attempting to add auto tree supports to overhangs it supports properly but adds random stray support layers out in thin air. These could obviously cause print failures due to excess filament being dragged onto layers or other supports. 

### Actual results

[image]

See attached screenshot with stray supports highlighted in red. 

### Expected results

Normal support generation without stray support material in thin air 

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12324368/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

whatthefilament, 2023-08-11:
I see others have submitted this issue. So closing as duplicate

SaltWei, 2023-08-15:
Has been fixed in internal version. Will be published soon

SaltWei, 2023-08-17:
Included in 1.7.4 which has been published. Please have a try! Thank you very much!

whatthefilament, 2023-08-17:
It seems to be generating support properly now 

Thanks!