bambulab/BambuStudio #2273: Conflicts of gcode paths when using arrange of multiple objects and rafts

fireplug911, 2023-08-11:
### Bambu Studio Version

01.06.01.00

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Will likely happen in a number of ways when multiple objects are on plate, but here is one way:
1. Add any small stl object to the plate.
2. select object and use clone to create 6 total objects.
3. right click on empty area of plate and select arrange
4. while in advanced mode: in prepare>support tab go to raft area and change to 3 raft layers
5. slice plate and error will occur 

### Actual results

When using the arrange tool it places the objects close enough together that if you then use the raft tool it will add conflicts to gcode paths.  I have confirmed in the actual print that the Bambu Lab X1C does indeed print the raft areas that overlap twice.

[image]
[image]

### Expected results

What should happen is either:

1. The arrange tool places objects far enough apart that subsequent use of a raft will not cause collisions. (adequate method, but then would lose build plate space.) or
2. When items are close together on the build plate and a raft is selected, the raft should encompass all items into one single raft instead of multiple individual rafts. (Best method used by other slicers that allow raft use.)

### Project file & Debug log uploads

[gcode conflict error.zip](https://github.com/bambulab/BambuStudio/files/12324509/gcode.conflict.error.zip)

[log.zip](https://github.com/bambulab/BambuStudio/files/12324578/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Chompworks, 2023-08-12:
> When items are close together on the build plate and a raft is selected, the raft should encompass all items into one single raft instead of multiple individual rafts. (Best method used by other slicers that allow raft use.)

This would definitely be preferable.

fireplug911, 2023-08-21:
Hmm.  I don't suppose there is any way to use the same code that can determine that there IS a conflicting area in the rafts to tell the slicer to give that area to either only one of the two rafts (which would essentially make one raft butt right up to but not conflict with the other raft) or simply delete that overlap area from BOTH rafts?

fireplug911, 2023-08-23:
Also, while on the raft subject, is there a reason why we have so few settings for rafts in Bamboo Studio to begin with?  I know in most cases rafts aren't necessary to get the desired print, but sometimes they are and the rafts produced in Cura are far superior in allowing ease of separation and smooth surfaces in the way those rafts are constructed, than the way Bamboo Studio constructs rafts.

[image]

joshmouch, 2023-10-13:
I just ran into this problem, too.

addictgamer, 2023-11-29:
Also running into this problem. I think my last slicer auto-merged rafts when they overlapped so I assumed the same would happen here, but that seems to not be the case (yet)?

ReinhardC, 2024-01-01:
Running into the same issue here. 

Couldn't the "object skipping" feature be separate from the raft building process?
Ie. make the raft a process of its own, and still have skippable objects on top?

Reality4DEvolution, 2024-02-02:
I FOUND A SOLUTOIN !!!!
if you have multiple small parts, or whatever, and do not care if you can't cancel a print (thus, the whole print job jmight have to be cancelled, BUT....)
add your parts to the bed in bambu studio.
SELECT them all
right click, and SAVE AS ONE STL.
this will save an STL (or 3mf) file with all these contained as ONE OBJECT.
clear off the plate
drag back in that STL or 3MF
it is ONE OBJECT.
set up the raft
no collisions occur.

NOTE, I am TESTNG THIS NOW, going to watch the printer see if it destroys itself....