bambulab/BambuStudio #2274: Issue with building on Ubuntu 20.04 with OCCT dependency

garnos, 2023-08-11:
### Bambu Studio Version

1.7.3

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 20.04

### Printer

N/A

### How to reproduce

I'm attempting to build the dependencies from source. I successfully ran this command from a fresh install of Ubuntu 20.04 after a `apt-get update` first:
`sudo apt-get install cmake git g++ build-essential libgl1-mesa-dev m4 libwayland-dev libxkbcommon-dev wayland-protocols extra-cmake-modules pkgconf libglu1-mesa-dev libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.0-dev libgstreamer1.0-dev libgstreamer-plugins-good1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad libosmesa6-dev`

The cmake command runs fine as well as: `cmake ../ -DDESTDIR="/opt/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1 &> ../../bambulogs.txt`

After that `make -j8 &>> ../../bambulogs.txt` runs into a build error with OCCT.
I'm wondering how to get around this one.

Snippet of the error here: 
```
Scanning dependencies of target tjunittest-static
Scanning dependencies of target tjbench-static
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 95%] Building C object CMakeFiles/tjbench-static.dir/tjbench.c.o
[ 96%] Building C object CMakeFiles/tjunittest-static.dir/tjunittest.c.o
[ 91%] Built target qdelaunay
[ 96%] Building C object CMakeFiles/tjunittest-static.dir/tjutil.c.o
[ 60%] Performing patch step for 'dep_OCCT'
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
make[2]: *** [CMakeFiles/dep_OCCT.dir/build.make:103: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch] Error 1
make[1]: *** [CMakeFiles/Makefile2:500: CMakeFiles/dep_OCCT.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Building C object CMakeFiles/tjunittest-static.dir/md5/md5.c.o
[ 92%] Built target rbox
[ 93%] Built target qconvex
[ 95%] Built target qvoronoi
[ 98%] Building C object CMakeFiles/tjunittest-static.dir/md5/md5hl.c.o
[ 96%] Built target qhalf
[100%] Built target testqset
[ 98%] Building C object CMakeFiles/tjbench-static.dir/tjutil.c.o
```

### Actual results

NA

### Expected results

NA

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

garnos, 2023-08-11:
Full logs here after running `make -j8 &>> ../../bambulogs.txt`:
[bambulogs.txt](https://github.com/bambulab/BambuStudio/files/12324728/bambulogs.txt)

blathers123, 2023-08-13:
I also ran into problems while building on Ubuntu one or two days ago, with similar issues to yours. I deleted the build directory and ran the BuildLinux.sh script which worked. Not sure why and it shouldn't be necessary, but it is a work around

watermanxxx, 2023-08-13:
When I build the code with Archlinux，I got similar error today
[image]

garnos, 2023-08-14:
> I also ran into problems while building on Ubuntu one or two days ago, with similar issues to yours. I deleted the build directory and ran the BuildLinux.sh script which worked. Not sure why and it shouldn't be necessary, but it is a work around

Running BuildLinux.sh just built the dependencies for you?

blathers123, 2023-08-14:
BuildLinux.sh checks that the dependencies are installed, and then does the full build. 

To get it to successfully build I:
1. Installed all the dependencies listed in the build for Linux guide 
2. Installed clang ("apt install clang" - I believe this is a missing dependency based on some errors I was getting without it
3. Deleted the build directory 
4. Ran "sudo .\BuildLinux.sh -u" to check for missing dependencies, needs to be sudo because it will install them for you. 
5. Ran "sudo .\BuildLinux.sh -dsi" to finish the build. I believe this needs to be sudo because number 4 needed sudo and there were some permission issues because of that

garnos, 2023-08-14:
> BuildLinux.sh checks that the dependencies are installed, and then does the full build.
>
> To get it to successfully build I:
>
> 1. Installed all the dependencies listed in the build for Linux guide
> 2. [...]

This worked for me and I was able to build. Thank you very much.

blathers123, 2023-08-14:
Glad I could help! 
I also tried to create a flatpak based off of the prusaslicer manifest but I'm limited by my cmake knowledge and wasn't able to get it to build successfully

closed 2023-08-29 by garnos