bambulab/BambuStudio #2275: Studio doesnt use correct color/slot when painting

Edzi11, 2023-08-11:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Open 3mf
2. Slice
3. top model has been painted with a yellow bandana while the bottom has a brown bandana
4. top model doesnt use the color that was used with slot one even if you change the color to any color as long as its from slot one / color one after slicing the model wont reflect that. 
[image]

### Actual results

top model with yellow bandana doesnt slice correctly / slot 1 color isnt being used

brown color doesnt have any issue being sliced 
[image]

### Expected results

top model should be using the color that was used to paint it (yellow) in slot 1 

### Project file & Debug log uploads

[nameplate example.zip](https://github.com/bambulab/BambuStudio/files/12325131/nameplate.example.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hcona, 2023-08-12:
I am having similar issues. When coloring one part of a model that is supposed to be extruded through the entire piece, only one side gets colored and the other side does not - I have to fill on the other side.  When slicing, it shows that the color is only a couple layers deep and does not go all the way through.  

Additionally, when I have more than one model to color, it merges the two items so I cannot color them separately.  I have to go to the object window and select each part separately, each color separately, in order to color the second item.

zhimin-zeng-bambulab, 2023-08-18:
When there are multiple parts of different colors under an Object, and they overlap each other, the material used for the overlapping part depends on the position of the part in the object list. The lower the position, the higher the printing priority.

So, if you want the correct color printing order, you can drag "Component 8" before the yellow Parts, the drag and drop function has been released in version v1.7.4, you can have a try.

[image]

hcona, 2023-08-18:
Thank you for the suggestion.  I was not aware of the drag-and-drop feature, but I have been able to manipulate the colors through the component boxes.

The specific issue I am having is this:  I am trying to test out a couple different prints, so I put one version in a file, colorize it, then import a different version of the file.  When I try to colorize it, it merges with the existing one even though I have selected only the new part, so I have to edit the colors in the component window.  

[image]

[image]

Edzi11, 2023-08-18:
> @Edzi11 When there are multiple parts of different colors under an Object, and they overlap each other, the material used for the overlapping part depends on the position of the part in the object list. The lower the position, the higher the printing priority.
> 
> So, if you want the correct color printing order, you can drag "Component 8" before the yellow Parts, the drag and drop function has been released in version v1.7.4, you can have a try.
> 
> [image]

Thank you, I'll give this a try and see if it helps!

zhimin-zeng-bambulab, 2023-08-30:
@Edzi11   Is this problem been resolved?

Edzi11, 2023-08-30:
> @Edzi11 Is this problem been resolved?

Yes, your Instructions worked thanks!

hcona, 2023-08-30:
Not mine.  When I have more than one model on a plate at a time, they
overlap each other and I cannot color them using the fill or the assembly
feature.  I have to go into each object, select each section, then
selecting the colors.

zhimin-zeng-bambulab, 2023-08-30:
@hcona  Could you share your 3mf file? And describe your expected results.

zhimin-zeng-bambulab, 2023-08-31:
@hcona  You can compress the 3mf file into a zip file and upload it.

zhimin-zeng-bambulab, 2023-08-31:
You can also tell me the BUG phenomenon you encountered and the expected result.

hcona, 2023-08-31:
[Problem Test.zip](https://github.com/bambulab/BambuStudio/files/12481992/Problem.Test.zip)
Here is the file.  

I am expecting to be able to place two different models on the same plate.  I expect to be able to select each model individually and color them separately.  Currently, when I have two models and I try to color them, they are stacked on top of each other and I cannot color them using the tools available.  They must be colored using the object window that lists all the parts.  I end up having to select each part, use the drop-down menu to select the color, then repeat this for every part on the model.

I have tried coloring one model, then importing the next model, but they get merged and the second model cannot be colored in the same manner.

Thanks for your help.

zhimin-zeng-bambulab, 2023-09-01:
[image]
@hcona  Even if they are stacked on top of each other, they can be colored, and you can choose different tools according to your needs. Could you have a try?

hcona, 2023-09-01:
Thank you, I have tried that.  The problem is that they are not really stacked on top of each other, they are merged, so I cannot separate the part to color.  I redid the example so you can better see the problem.  Let me know if this is a better example.  When you go to the assembly coloring, they are merged so that it is very difficult to choose the part I want to color.
[Problem Test (2).zip](https://github.com/bambulab/BambuStudio/files/12499842/Problem.Test.2.zip)

zhimin-zeng-bambulab, 2023-09-05:
@hcona 
I may understand what you mean. You want to select a part separately instead of the entire object for coloring. But currently we do not support selecting a part separately for coloring. You can create a new issue to record your request.

hcona, 2023-09-05:
Thank you, I will do that.

zhimin-zeng-bambulab, 2023-09-06:
Create a new issue  #2464 to track the issue of Select one part of the assembly individually for painting.

closed 2023-09-06 by zhimin-zeng-bambulab