bambulab/BambuStudio #2276: Linux Appimages don't Work in LAN mode

PRouzeau, 2023-08-11:
### Bambu Studio Version

All tested

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Additional system information

Intel i5-9500   8Gb RAM

### Printer

X1c

### How to reproduce

Start Appimage, click tab [Device] , click [No printer], which open a Menu where I find my Printer ('3DP-00M-837'  with bubble comment 'online'), then click on it. 

### Actual results

Error message: "Connect 3DP-00M-837 failed! [SN:xxxx, code=-1]", xxxx being the serial number of my machine.

A second attempt request the code (the first one did not), but after entering the code and confirming, I get the same message.
Tried to change the machine code, with same result.

### Expected results

Connection to my printer.

I tested a lot of Appimage (not only the official releases) with always the same result, except for an old one. 
The only Appimage which works is one which was published long time ago to show a Linux version working with the X1C camera. Unfortunately I was not able to find the post (this was on this github) where I found the link to this appimage and who published it, though I found the upload site here (I checked this is the same file as the one I use):
https://www.file-upload.net/download-15076245/BambuStudio_ubu64_2022-12-21_v1.4_x1camera.AppImage.html

This version works for me but it is quite old and does not work well with polygons, which is unfortunate for me as I only use polygons (modelling on OpenSCAD).

### Project file & Debug log uploads

[debug_Fri_Aug_11_23_05_58_25878.log.zip](https://github.com/bambulab/BambuStudio/files/12325350/debug_Fri_Aug_11_23_05_58_25878.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-08-19:
@PRouzeau 

does this issue still exists with the newest version 1.7.4?

and are you using the access code and ip from the one shown on your printer screen?

PRouzeau, 2023-08-19:
Yes, it still exists in 1.07.04.52. 
I use the access code indicated on my printer and I tried to change this code it to see if that was the problem, with the same result. As for the IP, This is not asked by the application and it did find the printer automatically.
The new code still works with the special version of 21 dec 2022 and at first start when the code is wrong, I get a message windows 'Wrong code', then with new code, the app does properly connect.

lanewei120, 2023-08-19:
> Yes, it still exists in 1.07.04.52. I use the access code indicated on my printer and I tried to change this code it to see if that was the problem, with the same result. As for the IP, This is not [...]

could you upload the new log files here?
including these two files
[image]

before upload this file, just try the steps and close bambu studio

@walterwongbbl will have a check

PRouzeau, 2023-08-19:
Here is the network log. Seems not capable to upload another file, I will attempt in an other message
[debug_network_Sat_Aug_19_11_40_06.log](https://github.com/bambulab/BambuStudio/files/12385612/debug_network_Sat_Aug_19_11_40_06.log)

Please also note that During the attempted connection, I see very fugitively a screen with the printer and what seems to be its serial number, I am trying to record that.

PRouzeau, 2023-08-19:
[PRZ_Bambu_logs_19_Aout_23.zip](https://github.com/bambulab/BambuStudio/files/12385623/PRZ_Bambu_logs_19_Aout_23.zip)
Here are both files, I forgot that I have to zip them before

walterwongbbl, 2023-09-18:
HI.  @PRouzeau 
I checked the LOG file you provided and found that the first connection attempt failed because the access code was incorrect.
[image]

You can try deleting the configuration file and then running Studio again.
[image]

PRouzeau, 2023-09-18:
I deleted both configuration files (bambu studio and Network) and it doesn't changed anything. As I already wrote, on first connection trial, I got directly the error message but at the second trial (in the same session), I am requested to enter the code, then after entering the code I got the same error message, this is always the same. Please note that I tested with new codes with always the same result.
Please find attached the log files
[Bambu_log_files_18_11.zip](https://github.com/bambulab/BambuStudio/files/12648461/Bambu_log_files_18_11.zip)

walterwongbbl, 2023-10-13:
Hi. @PRouzeau 
Please download and use the latest version of Studio

walterwongbbl, 2023-12-21:
Hello, we are very sorry to inform you that we are busy with the release of a new version. May I ask if your issue has been resolved

closed 2024-01-18 by walterwongbbl