bambulab/BambuStudio #2279: Flow Dynamics Calibration

sibor64, 2023-08-12:
Good morning
When you want to edit or delete a reference
nothing is saved
Have a good day
Boris
[image]

k8stuff, 2023-08-12:
I have the same. I printed calibrations for 8 spools across two AMS and named the filaments with their respective K-Values. 
Today I noticed that all K values are set to 0 but I was able to choose the K profile created again. 

Earlier I swapped a nozzle hotend and pulled the power from the printer and now I can see that the k value profiles are gone too.

zhimin-zeng-bambulab, 2023-10-11:
@sibor64  Could you describe your requirement in more detail?