bambulab/BambuStudio #2280: Flow Dynamics Calibration

RaffaelGitHub, 2023-08-12:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Additional system information

CPU: Apple M1 Pro
macOS: 13.4.1 (22F82)

### Printer

Bambu Lab P1P

### How to reproduce

1. Open Bambu Studio
2. Press "Calibration" tab
3. Press "Manual Calibration"
4. Choose "Plate Type" -> "Textured PEI Plate"
5. Choose the right "Filament For Calibration"
6. Press "Calibrate"

### Actual results

Nothing will happen 

### Expected results

Flow Calibration

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12327938/log.zip)
[HoneyComb Wall inserts_containers_holders.3mf.zip](https://github.com/bambulab/BambuStudio/files/12327941/HoneyComb.Wall.inserts_containers_holders.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

wiz78, 2023-08-26:
I have the same problem. Can't start manual calibration...

ma-gu-16, 2023-09-22:
Same for me on Windows 11 pro

zhimin-zeng-bambulab, 2023-09-25:
Could you have a try in V1.7.6.  This version has some modifications related to Flow Dynamics Calibration.

zhimin-zeng-bambulab, 2023-10-11:
@RaffaelGitHub Does this problem still exist?

closed 2023-10-13 by zhimin-zeng-bambulab