bambulab/BambuStudio #2282: Build plate disappears

ecoughtry, 2023-08-12:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Ventura 13.5

### Additional system information

AMD Radeon Pro 5300M 4 GB
Intel UHD Graphics 630 1536 MB
Memory: 32GB DDR4

### Printer

x1c

### How to reproduce

1. Open app
2. No build plate, even after import or "slice" it will not visualize what should be there

### Actual results

No build plate

### Expected results

Visual build plate displayed

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/12328362/Archive.zip)
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-08-18:
Hi. @ecoughtry 
Can't  display the build plate every time？

StoneLiBambulab, 2023-08-18:
hi @ecoughtry ,

your opengl version is 2.0, please try to upgrade your video driver, update the opengl version to 4.x

Thanks!

Stone Li

Haidiye00, 2023-09-04:
Hello, have you upgraded the opengl version and resolved your issue ?

Haidiye01, 2023-12-18:
Hello, have you upgraded the opengl version and resolved your issue ?

SaltWei, 2024-01-16:
No update from user for a long time. Close.
Reopen if has new response.

closed 2024-01-16 by SaltWei