bambulab/BambuStudio #2285: Allow skipping of individual objects in an assembly

kevinclark, 2023-08-13:
**Is your feature request related to a problem? Please describe.**

Today I have a failed print where one object that had been grouped into an assembly of 4 failed, but I can only skip the entire assembly.

**Describe the solution you'd like**

I’d like to be able to select individual components of an assembly. 

**Describe alternatives you've considered**

It seems possible that if I had ungrouped the objects after layout I could still skip the one item.

kevinclark, 2023-08-13:
I suppose another alternative would be to allow a setting where the assemblies are always split into individual options when sliced?

SaltWei, 2023-08-16:
Currently the min skipping unit is object. We will consider your suggestion carefully. Thank you very much!

kevinclark, 2023-08-16:
Thank you! For consideration, the motivating example was the [pastamatic winder](https://www.printables.com/model/466883-pastamatic-filament-spool-winder-for-bambu-lab-x1c) which distributes its project as a 3mf. For whatever reason they group some sets of objects into assemblies.