bambulab/BambuStudio #2288: P1S Freezing and restart when trying to run Flow Dynamics calibration

Waynegodli, 2023-08-13:
### Bambu Studio Version

v01.07.03.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Labs P1S

### How to reproduce

1. Go to Calibration
2. Select Flow Dynamics
3.  Manual Calibration
4. Select Plate type, AMS Slot and Filament Profile.
5. Click calibrate

### Actual results

Print freezes, screen goes black and restarts itself..... doesn't print... Fine sending other files to print

### Expected results

Dynamic Flow Calibration should of started print

### Project file & Debug log uploads

[debug_network_Sun_Aug_13_18_25_40.log](https://github.com/bambulab/BambuStudio/files/12330039/debug_network_Sun_Aug_13_18_25_40.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

byWulf, 2023-08-22:
Happens to me too. I used the following settings:
* 0.4 nozzle
* High Temp Plate
* External Filament (not AMS): Generic PLA

I have a P1P with the latest firmware 01.04.00.00 and the latest Bambu Studio (1.7.4.52).

My printer just freezes (controlling the display is only barely possible, it is very very laggy and react veeery slow). I had to shut it off and on again to become normal again.

Thank you for looking into the problem!

PartyLime99, 2023-09-18:
Same problem for me. I can't seem to send any of the calibrations. I have the latest firmware and software installed

sinkewa, 2023-09-19:
Exactly the same here. P1S with AMS

sinkewa, 2023-09-19:
> Exactly the same here. P1S with AMS FW 01.04.00.00 Bambu Studio 1.7.4.52

Tested in Orca slicer, works just fine, so not a firmware issue.

PartyLime99, 2023-09-19:
I tried again and got the same N/A message and the printer seemed to drop off the network. I left it for a while though and it finally started and ran with no issues

arnrist, 2023-09-22:
Same issue with the P1P. Been occuring for a while.
After making selection in the Preset tab, and goes to Calibration with status N/A, the P1P restarts it self.
On my latest try, the calibration finally went through after 11th (or so) try.

0.4 nozzle
Textured plate
Generic PETG profile
AMS slot 4

P1P FW v 01.04.00.00
AMS FW v 00.00.06.32
Bambu Studio 1.7.4.52

sinkewa, 2023-09-22:
I've noticed that the calibration does start successfully if I had run an extrusion calibration just prior to it. Might be something to do with the homing status or nozzle / bed temp. Possibly some error in g-code that causes the MCU to freeze and reboot?

zhimin-zeng-bambulab, 2023-09-25:
This BUG has been fixed in V1.7.6, you can have a try.

zhimin-zeng-bambulab, 2023-10-11:
@Waynegodli Does this problem still exist?

closed 2023-10-13 by zhimin-zeng-bambulab