bambulab/BambuStudio #2289: Slicer states "no layers detected" when i place anything on plate 1 - all parts work on all other plates

preeny91, 2023-08-13:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

bambu lab p1p

### How to reproduce

Move any of the parts to plate 1 and slice, error code then appears

### Actual results

shows error "no layers found"

### Expected results

if the parts are moved to any other plate other than plate one, then they will slice 

### Project file & Debug log uploads

[full mask project bug.zip](https://github.com/bambulab/BambuStudio/files/12330111/full.mask.project.bug.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12330112/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-08-15:
[image]
what the stl is?

QingZhangBambu, 2023-08-15:
[image]
this very small part produce empty layer.

closed 2023-10-16 by QingZhangBambu