bambulab/BambuStudio #2292: Bambu Studio

jtusman, 2023-08-13:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS  12.6.8 (21G725)

### Printer

X1 Carbon

### How to reproduce

When the print is complete and try to send another print to the X1 it will not send due to Bambu Studio saying the printer is still busy. Bambu Studio shows the printer has about 30% to go until finished.

### Actual results

I have to sign out of the software and sign back in to be able to start the next print.

### Expected results

When the print completes it should let me send the next print as it has up until this new version.

### Project file & Debug log uploads

[Uploading log.zip…]()

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-10-13:
Hi. @jtusman 
We realize this might be an issue, where the printer status didn't sync successfully. Have you ever put your computer to sleep?

walterwongbbl, 2023-12-21:
If you have any other issues, please create a new issue. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl