bambulab/BambuStudio #2294: Slicing Error with AMS: "WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect."

thereseLYR, 2023-08-14:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 Home 22H2

### Additional system information

CPU: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
Memory: 32GB
Graphics Card: NVIDIA GeForce RTX 2060

### Printer

Bambu Lab P1P

### How to reproduce

1. Open 3MF file
2. Slice plate 01 (Volcano Base Main Middle)
3. Error regarding 'WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect.' will be thrown

### Actual results

[image]
Error is thrown, and slicing is not completed

### Expected results

Expected slicing to be able to complete.

### Project file & Debug log uploads

[Volcano Center Piece_troubleshooting.zip](https://github.com/bambulab/BambuStudio/files/12331091/Volcano.Center.Piece_troubleshooting.zip)
[logs_14Aug.zip](https://github.com/bambulab/BambuStudio/files/12331097/logs_14Aug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

thereseLYR, 2023-08-14:
This bug might be related to #1568 and https://github.com/prusa3d/PrusaSlicer/issues/3125, but is still present in the latest release of Bambu Studio (1.7.3.50)

zhimin-zeng-bambulab, 2023-10-24:
we will fix it in 1.8 beta

zhimin-zeng-bambulab, 2023-11-13:
It has been fixed in v01.08.00.57, you can have a try.

closed 2023-11-13 by zhimin-zeng-bambulab

ash286, 2024-11-07:
Has this bug fixed in latest v2.2.0 official? I'm keep encountering with this error of "Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect."