bambulab/BambuStudio #2296: Flow Rate Calibration Pattern too crowded with 0.8mm nozzle

qeggs, 2023-08-14:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Not Applicable

### Additional system information

Not Applicable

### Printer

X1C

### How to reproduce

1. Install 0.8mm nozzle from Bambu Lab.
2. Run the Manual Flow Rate Calibration.
3. Observe that the array of calibration objects are overlapping.

Note: Orcaslicer has this same issue, but in Orcaslicer, the objects can be manually spaced apart to avoid overlapping. So there is a workaround in Orcaslicer but no workaround available in Bambu Studio.

### Actual results

The tabs with the flow rate numbers on them are halfway overlapping the rectangular test regions of the adjacent objects. See photo.

[image]

### Expected results

Array of test objects should be space out such that there is no overlap between objects when using a larger nozzle such as the 0.8mm nozzle.

### Project file & Debug log uploads

Not Applicable, this bug occurs during Flow Rate calibration (manual).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-08-16:
Thanks for report this issue, we will fix it the next version.

zhimin-zeng-bambulab, 2023-08-17:
It has been fixed in 01.07.04.52

SaltWei, 2023-08-17:
Fixed in latest 1.7.4

closed 2023-08-17 by SaltWei