bambulab/BambuStudio #2298: flow dynamics Mac Os

MarcoPiazza83, 2023-08-14:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

13.5 (22G74)

### Additional system information

Mac Os Intel 

### Printer

bambu lab x1C

### How to reproduce

Flow dynamics
manage result

### Actual results

no save data

https://github.com/bambulab/BambuStudio/assets/4783556/2920db44-4cbc-4335-ad46-254bacad3d47

### Expected results

save data

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12337973/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-08-16:
@MarcoPiazza83 
Could you switch the language to English to see if this problem still exists?

SaltWei, 2023-08-16:
Maybe similar to #2141

MarcoPiazza83, 2023-08-16:
@SaltWei @zhimin-zeng-bambulab

Yes, in other language the problem is resolve.
Data has been saved

SaltWei, 2023-08-17:
Fixed in latest 1.7.4

closed 2023-08-17 by SaltWei