bambulab/BambuStudio #2299: Calibration for non-bambu printers

Jake-N-Bake, 2023-08-15:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 22H2

### Printer

Bambulab X1 Carbon Combo

### How to reproduce

1. Load an Elegoo Neptune 3 Pro printer on the slicer.
2. Insert one of the flow rate calibrations onto the build plate.
3. If I click anywhere else Bambu Studio will crash completely or if I slice the flow rate test then go back to the prepare tab, it will crash completey.

### Actual results

Bambu Studio will completely crash without warning. Sometimes it will throw an error then crash.

### Expected results

I should be able to mess with the settings and be able to click on other parts of the build plate without the program crashing.

### Project file & Debug log uploads

There is no project file, Its the integrated calibration prints for non Bambu Printers.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-08-16:
could you have a try for this [image ](https://github.com/bambulab/BambuStudio/releases/download/v01.07.03.50/Bambu_Studio_win_public-v01.07.03.50-20230810190205.exe)?

zhimin-zeng-bambulab, 2023-08-17:
It has been fixed in 01.07.04.52