bambulab/BambuStudio #2301: Printer Won't Connect Wirelessly To My Computer/Prints Won't Send To Printer

LaZyPrintz, 2023-08-15:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.42 GHz, 8.00GB RAM
Galaxy Notebook

### Printer

Bambu Lab P1S

### How to reproduce

1. Open Bambu Studios
2. Slice the print.
3. Click on Print
4. Send to printer
5. That's when the issue occurs.
6. Gives error code (-3080)
7. It says to check the current status of the Bambu Server.

### Actual results

I'm not able to send the Gcode files from the computer to the printer wirelessly.

### Expected results

The Gcode should be sent straight to the printer and start printing once it downloads it.

### Project file & Debug log uploads

"C:\Users\justc\Downloads\Stormbreaker-Handle-V2-002.zip"

### Checklist of files to include

- [ ] Log file
- [X] Project file

walterwongbbl, 2023-08-18:
Hi. @LaZyPrintz 
At that time, there were some issues with our server. 
Can you use a printer for printing now?

walterwongbbl, 2023-09-07:
@LaZyPrintz 
Is everything okay now？

LaZyPrintz, 2023-09-20:
It’s working now when you fixed the cloud. Apparently it went out for a lot
of users. Thank you!

———————————

Cody S. Inbody
Mobile - (918) 809-9979
3913 N. 33rd Pl.
Broken Arrow, OK 74014