bambulab/BambuStudio #2302: Issues sending to print

yellownick, 2023-08-15:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1P

### How to reproduce

Slice then Print plate through cloud service

### Actual results

file couldnt be sent, Error code -3040, -3060, -3080

### Expected results

file sent to printer

### Project file & Debug log uploads

[debug_network_Tue_Aug_15_13_07_02.log](https://github.com/bambulab/BambuStudio/files/12343871/debug_network_Tue_Aug_15_13_07_02.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-08-18:
@yellownick 
At that time, there were some issues with our server.
Can you use a printer for printing now?

closed 2023-10-13 by walterwongbbl

MCLARKIND, 2024-08-15:
@walterwongbbl   I am having this same problem.  When I test services I get this message.
Thu Aug 15 13:49:18:test storage start...
Thu Aug 15 13:49:18:[test_oss]: url=http://upload-file.bambulab.com
Thu Aug 15 13:49:20:test storage failed
Thu Aug 15 13:49:20:status=404, body={"error_msg":"404 Route Not Found"}
, error=

Nico04, 2024-10-02:
Same issue here, even after full network config reset :(