bambulab/BambuStudio #2305: Proper handling of a cloud outage including clearing the queue before it's brought back up.  Also safeguards between jobs

sprior, 2023-08-15:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Attempt to start a print, it says success but the printer doesn't start printing.
2. Try again, possibly more than once.
3. Finally realize there is a cloud issue and stop
4. Some time later the printer just starts printing and when it's finished with the first job it proceeds to try to print the next one.

### Actual results

When there's a Bambu cloud issue it is common for the user to not realize it and resubmit a print job to try again.
It is also common that when the cloud service is brought back up (when the user is not necessarily expecting it) that the printer starts printing on its own and worse starts printing a new job while the previous one is still on the build plate - this could result in hardware damage.

### Expected results

There should be a better defined procedure taken by Bambu cloud staff to bring up the system after an outage.
First is that the job queue should be cleared before the cloud service is restarted.
There is currently no way for a user to view or clear the cloud job queue for their printer, they just have to be there to cancel it when a job starts - not good.
I understand that a queue exists just as a way of implementing the cloud service, but I question whether a printer allowing multiple jobs on the queue even makes sense since the printer needs to be physically cleared between jobs.
I think when a new job is submitted for a printer it should replace any non-started job for that printer already in the queue.
If you are determined to keep multiple jobs in the queue for a printer, then a safety measure such as requiring the door to open and close before the next job is allowed to start should be implemented.

### Project file & Debug log uploads

[cube.zip](https://github.com/bambulab/BambuStudio/files/12352308/cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

jason-e-gross, 2023-08-23:
I'd read the blog post -- and not so sure a time-interval is a solid approach.     What about using idempotency, and give the specific print job a unique ID.   If the printer receives the same unique ID after having already received it - ignore the second send.

sprior, 2023-08-23:
I don't see where I mentioned a time interval at all.  As far as I'm concerned before they bring the cloud server back up they should purge the entire print queue as long as they don't kill anything that's already printing.  The only other reference I made was that a new job should replace any existing one in the queue for that printer, but that's based on the queue being an ordered collection, not time.  I also don't agree with your suggestion of ignoring a second send of the same id - you shouldn't allow a second print job to be queued to the same printer even if they are completely different jobs with different id's.  I don't think that having a queue for a printer makes physical sense when manual intervention is needed between jobs to clear it.