bambulab/BambuStudio #2310: [Feature Request] - Bedmesh information

SnickCH, 2023-08-16:
**Is your feature request related to a problem? Please describe.**
As far as I can tell, there are many people that have warped beds or are not happy with their bed. At the moment there is no way to manually adjust anything (like with tape etc.) because your customers don't know where the bed is uneven.

We need this data to know were we have to adjust it. Or a second case if the bed-tramming was not done correctly from the factory (QC). With a simple bed mesh and the values we can see if this is the case.

**Describe the solution you'd like**
General a way to show the values from the bedleveling and optionaly (for later) a visualized bedmesch.

**Describe alternatives you've considered**
Manuall measurements etc. that are unprecise.

**Additional context**
The only reason NOT to give us this data would be, that bambulab has to many warped beds (even for customer that don't complain because they have only decoration things) and you want to hide it. If your product is decent you have nothing to worry about it if you release this data.

JSorenSwanson, 2023-08-27:
There's no good reason to withhold this information. If you want your IP to be completely proprietary, fine – allow us the instrumentation. 

This is a bad look all around

SnickCH, 2023-08-27:
@JSorenSwanson The only problem I have with the printer is the warped printbed - that support says it is not warped (of course) but they are still sending me a replacement. 

I received the printer 4-5weeks ago and since then it is more or less unusable because of this. With the bedmesh information I could easily fix it myself - then I would know where the problem is.