bambulab/BambuStudio #2311: Can't Close Filament Windows

mltriebe, 2023-08-16:
### Bambu Studio Version

v01.07.04.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel HD Graphics 620 GPU, at 1366x768 resolution

### Printer

Bambu Lab X1C

### How to reproduce

1. Open Studio to the Device tab.
2. Click on a filament to select a PA profile or any change.

### Actual results

I can not click on the filament window to close it. Therefore I can not select anything or move the entire window to close it. I have had to use taskmanager 2 times so far to close Bambu Studio.

### Expected results

I should be able to close the filament window to continue in Bambu Studio.

[image]

### Project file & Debug log uploads

There is no log file due to me having to close the program through taskmanager.

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-08-18:
Hi.  @mltriebe 
What is the resolution of your monitor?

mltriebe, 2023-08-18:
As stated above:

Additional system information
Intel HD Graphics 620 GPU, at 1366x768 resolution

walterwongbbl, 2023-08-18:
@mltriebe 
Is the maximum resolution you support 1366x768?

mltriebe, 2023-08-18:
Yes, it is the max.

Mike

bam-winterfell, 2023-08-29:
Hello, we have adjusted the position of this window and it will be launched in the next version. Please update it promptly.
Wishing you a pleasant life

walterwongbbl, 2023-12-21:
If you have any other doubts, please create a new issue. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl