bambulab/BambuStudio #2312: Nozzle 0.6mm is not supported for auto calibration

Datt-Flo, 2023-08-16:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Install 0.6mm Nozzle
2. try to auto calibrate

### Actual results

[image]
[image]
[image]
[image]

### Expected results

Running Test

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mltriebe, 2023-08-16:
It only supports a .4 nozzle for now per this wiki page.  https://wiki.bambulab.com/en/software/bambu-studio/calibration_flow_rate

SaltWei, 2023-08-17:
Feature under developing, not bug

Datt-Flo, 2023-08-17:
> Feature under developing, not bug

Ok

SnickCH, 2023-08-17:
Interesting. Just last weekend I did an autocalibration for 3 AMS slots (at the same time) with my 0.6mm nozzle. I just retested it - I can start this calibration. My UI also looks different then the picture of @Datt-Flo 

Version: 1.7.4.52