bambulab/BambuStudio #2315: Linux Mint, program crash whrn playing stream p1p

Berden2162, 2023-08-17:
### Bambu Studio Version

1.7.4

### Where is the application from?

Bambu Lab github releases

### OS version

linux mint 

### Printer

P1S

### How to reproduce

click play icon 

### Actual results

program crash

### Expected results

play stream

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-08-17:
are you using the fedora image?

we just uploaded the ubuntu image https://github.com/bambulab/BambuStudio/releases/download/v01.07.04.52/Bambu_Studio_linux_ubuntu-v01.07.04.52.AppImage

how about this one?

Berden2162, 2023-08-17:
Good, work well, Thank you

closed 2023-08-18 by lanewei120