bambulab/BambuStudio #2317: Request for Enhanced Control and Default Orientation Preservation in Bambu Studio CLI

Atanas-Kolev, 2023-08-17:
**Description:**
We have recently purchased two Bambu printers and are in the process of integrating them into our automation farm. As part of this integration, we are considering the use of Bambu Studio CLI within our software pipeline. However, we have specific requirements regarding object rotation, placement on the build plate, and the preservation of the default orientation that we would like to address.

**Integration Goals:**
Our objective is to optimize our 3D printing processes using Bambu Studio's CLI capabilities. To align with our production needs, we need enhanced control over rotation and object placement on the build plate.

**Specific Requirements:**

1. **Rotation Control:** We need the ability to define custom rotation angles for the 3D models processed by Bambu Studio to ensure optimal printing orientations based on our design considerations.

2. **Placement on Build Plate:** Precise object placement on the build plate is crucial for efficiency. We require a feature to input coordinates or align objects consistently.

**Current Behavior and Orientation Preservation:**
Currently, the output is auto-oriented by Bambu Studio CLI regardless of the input STL file orientation. We aim to preserve the default orientation of STL files throughout processing to maintain compatibility with other software tools and our existing workflows.

**Feature Enhancement Request:**
We request support for these requirements. If features are available or alternative approaches exist, guidance on achieving these goals would be appreciated.

Additionally, we express our interest in including these capabilities in future Bambu Studio CLI updates, especially for users involved in industrial and automated 3D printing.

lanewei120, 2023-08-21:
@Atanas-Kolev 

we can improve the logic,
however we can not ensure the success probability of this print

how about load these stls into 3mf, do the oriant and arrange, add some paramters if needed, and then pass the 3mf to CLI for your cases?

Atanas-Kolev, 2023-08-21:
@lanewei120
Hi, I need to be able to do all the things from the CLI or by Python script. I tried to rotate the STL and then import it, but the same effect occurs (it gets autorotated).

Atanas-Kolev, 2023-08-21:
@lanewei120 
i need comands like --rotate-x,  --rotate-y and  --rotate-z

lanewei120, 2023-08-21:
doing rotate in the command Line can not ensure the print successfully
I suggested to do it in Bambu Studio and export it to a 3mf

we can pass 3mf to CLi directly.

Could you show us your detailed use case?

Atanas-Kolev, 2023-08-21:
@lanewei120 
We are implementing complete automation for the 3D printing process, and the command-line rotation will be determined based on the orientation provided by our clients. For more information contact me on atanas@dhr.is

Atanas-Kolev, 2023-08-22:
@lanewei120 
Could we include this feature? We're looking for either a rotation function or a method to deactivate automatic orientation. This way, we can simply position it on the build plate with the default orientation from the STL file.

lanewei120, 2023-08-22:
> @lanewei120 Could we include this feature? We're looking for either a rotation function or a method to deactivate automatic orientation. This way, we can simply position it on the build plate with the default orientation from the STL file.

I will send you a email for the detailed requirement

closed 2023-08-30 by Atanas-Kolev