bambulab/BambuStudio #2318: [Feature Request] - Skirt - Distance to object

SnickCH, 2023-08-17:
**Is your feature request related to a problem? Please describe.**
The Skirt distance is not far enough away from the object. If now a bigger "blob" of Filament gets stuck on the Skirt (this is why I use it mostly) it will stick to the print.

**Describe the solution you'd like**
A value "Skirt distance to object" that I can define any value (1 - 200mm). I often use for my Prusa 20-50mm distance, depending on the object size.

**Describe alternatives you've considered**
I have to manually remove the skirt before the print is at the same possision.

SaltWei, 2023-08-17:
OK. Accept. We will change the max limitation to be 50mm

SnickCH, 2023-08-17:
@SaltWei Thank you for the fast reply. Just to be sure: The Brim already has a value that we can define. The Skirt has no distance value at all (at the moment), there it is a fixed value (Skirtloops and Skirt height can be set, but no distance).

SaltWei, 2023-08-17:
[image]

SaltWei, 2023-08-17:
[image]
You can see more settings in developer mode

SaltWei, 2023-09-01:
So this ticket can be resolved and closed now?

SnickCH, 2023-09-01:
@SaltWei Thank you very much, it can be closed

closed 2023-09-01 by SnickCH

pedrokv, 2023-12-30:
> ![image](https://private-user-images.githubusercontent.com/110660407/261307795-9b664f2d-06f2-4c4c-b58e-d5ddabfeed36.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM5NTAyNjYsIm5iZiI6MTcwMzk0OTk2NiwicGF0aCI6Ii8xMTA2NjA0MDcvMjYxMzA3Nzk1LTliNjY0ZjJkLTA2ZjItNGM0Yy1iNThlLWQ1ZGRhYmZlZWQzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjMxMjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIzMFQxNTI2MDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMDdiMzM1MmJjMzdkMTc1N2E2N2E1NDQxYzY5N2YzNjUyYWY5YmRjODU5ZWQyM2UwY2JmNGNhNDlmZWFmNzQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QdFb6VZW_nPVyXdvCO05AypbXT-_06y_4UR6Trol-c4) You can see more settings in developer mode. These settings are for expert users or developers. Less used by newbie. But the skirt distance is limited to max 10 at the moment. We will change it to 50mm

I would like to ask for having it under "advanced settings", not under development mode. Skirt distance is nearly a basic settings in slicers, nothing that I would expect under development mode.