bambulab/BambuStudio #2319: Add firmware retraction support for third-party printers

meltiseugen, 2023-08-17:
**Is your feature request related to a problem? Please describe.**
I usually handle the retraction settings from the firmware (klipper). So I would like to have this feature in BambuStudio.
Mainly to convert the retraction moves to G10 and G11 gcode commands.

**Describe the solution you'd like**
A checkbox in the printer settings tab, to be able to enable/disable firmware retraction

**Describe alternatives you've considered**
A post-processing script that would replace all the retraction moves with G10/G11 commands

**Additional context**
You can see the same feature in OrcaSlicer, PrusaSlicer, SuperSlicer

SaltWei, 2023-08-18:
Accept. We will add it back for 3-rd printer

XunZhangBambu, 2024-01-05:
@meltiseugen  In latest version 1.8.2 Public Official , we have added this feature

closed 2024-01-16 by XunZhangBambu