bambulab/BambuStudio #2320: Log full 3D printing history in Bambu Studio

wouterfranssens, 2023-08-17:
The Bambu Handy app offers the possibility to show the printing history of a single user.
It would be a nice feature that there is also a way to download the full history of what has been printed (by all users).
This way, the used filament could be billed on different projects.
Contrary to the user printer history in the app, users should not be able to delete this full printing history.

chrisribe, 2023-10-04:
Yes what gives why can I not access this print history in bambu studio ?

joshmitcho, 2024-02-12:
Any updates on this? @lanewei120

SaltWei, 2024-02-20:
> It would be a nice feature that there is also a way to download the full history of what has been printed (by all users).
> This way, the used filament could be billed on different projects.

Is there some other user scenario? 
What kind of file format of downloaded history and what information should be necessary in it?
More suggestions would be nice for designing

joshmitcho, 2024-02-20:
@SaltWei I'd assume a csv would be sufficient, so it would open in Excel
Date, filename, user who printed it, amount and type of each filament used. That would be sufficient for anyone to calculate their own totals for cost recovery/billing.

Kayu84, 2024-06-14:
A history would be very helpful for various reasons:

- Projects that you do not own, have not printed: average filament used, time and cost calculation
- Projects that you own, have already printed: a nice chart of completed jobs, overview of reprints informations
- overall: also a nice history of the time you have spent printing, how much and which filament you have used...

But please don't add all users (data privacy), I have no idea where it is helpful to know who printed how much filament and how much time spent. Or just make it optional, users who want to share can set this as profile settings.