bambulab/BambuStudio #2321: Stay in status : Print in progess with 100% complete...

dguinaldo, 2023-08-17:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MACOS 13.5

### Additional system information

Stay in status : Print in progess with 100% complete...

The status is idle in the mobile application. But even after a full restart of the application, it keep in "Print n progress" on the MACOS version

### Printer

X1C

### How to reproduce

Execute a print
Wait for the print to finish
try to send a new print
The status in Bambu studio on MAC stay on "Print in progress" but the print is done on the printer and status is idle on the mobile app.

### Actual results

Unable to send a new print

### Expected results

The status of the printer should be "idle"

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12372075/log.zip)
[winder-cord-stackable-knurld.3mf.zip](https://github.com/bambulab/BambuStudio/files/12372078/winder-cord-stackable-knurld.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-08-18:
Hi. @dguinaldo  
What is the status on the printer screen?   "Print in progress" or completed?

walterwongbbl, 2023-10-13:
Hi. @dguinaldo 
Hello, we've identified this issue. Has your computer recently gone through a sleep mode?

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl