bambulab/BambuStudio #2322: Tree supports are generated with a layer shift

Hologos, 2023-08-17:
### Bambu Studio Version

1.7.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.5 (22G74)

### Printer

Bambu Lab P1P

### How to reproduce

1. Download attached 3mf.
2. Slice it.
3. Check generated supports. There is a layer shift in some of them.,

### Actual results

Some supports have a layer shift and it makes them weaker and more prone to breaking. I had failed print because of this.

[image]

### Expected results

No layer shift in generated supports.

### Project file & Debug log uploads

[pokemon charizard 0.12mm fixed z-hop.3mf.zip](https://github.com/bambulab/BambuStudio/files/12373327/pokemon.charizard.0.12mm.fixed.z-hop.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Hologos, 2023-08-17:
As expected the branch broke.. I had to glue it back.

[image]
[image]

Hologos, 2023-08-25:
@tangketan Hello, is this gonna be fixed in next release or will it take longer to fix? Thank you

tangketan, 2023-09-14:
@Hologos I'll fix it in the next release.

tangketan, 2023-12-21:
Offical 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan