bambulab/BambuStudio #2323: Nozzle distance

Paragaz, 2023-08-17:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C with AMS

### How to reproduce

1. Update Bambu Studio to the newest Version 1.7.4.52
2. Print with 0.2mm default Settings on cold Plate with Bambu basic PLA
3. The options print bed leveling, flow calibration, timer laps and AMS are selected 
4. Print doesn't adhere on the plate
5. Downgrade to version 1.7.3.50
6. Same Settings and model
7. The Print adhere to the bed 

### Actual results

[GCODE.zip](https://github.com/bambulab/BambuStudio/files/12373395/GCODE.zip)
I tried the Benchy from the internal storage and it was fine.
Then i sliced a Benchy with the newest version and the front doesn't stick to the bed.
The purge line and the flow calibration sticked optimal to the plate.

### Expected results

Has the nozzle spacing been changed in the new version?
I could not find an option for this in the settings.
How can I adjust it?

### Project file & Debug log uploads

[Klinge.zip](https://github.com/bambulab/BambuStudio/files/12373532/Klinge.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-08-18:
@QingZhangBambu Please find a 0.2 nozzle machine and have a check. I find no difference between there two version of studio which may cause this issue. Maybe just random adhesive issue

Paragaz, 2023-08-18:
@SaltWei Thanks for your quick response
I was again looking for a difference between the two versions.
In the automatic flow dynamics calibration, the K values are very different.
With the new version the determined K values are 0.008 and with the old version 0.017.

I have taken a closer look at the first layer of the parts that have come off.
I believe that my first assumption was wrong and I now believe that not enough material is squeezed out of the nozzle.

[image]