bambulab/BambuStudio #2324: Error on Step File Reload

stan-po, 2023-08-17:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Additional system information

Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz, 3600 Mhz, 8 Core(s), 8 Logical Processor(s)
32 GB RAM

### Printer

X1 Carbon

### How to reproduce

1. Insert STEP part
2. Make changes to STEP part in CAD
3. Click "Reload from Disk"
4. Reload is OK
5. But then shut down Bambu Studio and re-open project
6. Goto part and click "Reload from Disk"
7. Error is generated and part is not updated

### Actual results

[image]

### Expected results

Part should be updated

### Project file & Debug log uploads

[Test.zip](https://github.com/bambulab/BambuStudio/files/12374179/Test.zip)
[debug_network_Fri_Aug_18_09_33_45.log](https://github.com/bambulab/BambuStudio/files/12374183/debug_network_Fri_Aug_18_09_33_45.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-08-18:
@QingZhangBambu Please check whether can reproduce first

stan-po, 2023-08-21:
Yes, completely reproducible.
This error has been present since the beta release of 1.7.0

Regards,

Stan Poniger

DanBao-Bambu, 2024-01-16:
@stan-po Hello, please try using version v1.8. This issue should have been resolved.

SaltWei, 2024-01-18:
@stan-po Does this issue still exist on 1.8 under your PC environment？

stan-po, 2024-01-23:
Issue was solved in the very latest release 1.8.4.51
Stan Poniger

SaltWei, 2024-01-23:
> Issue was solved in the very latest release 1.8.4.51

Thanks for response!

closed 2024-01-23 by SaltWei