bambulab/BambuStudio #2325: default name gcode/print files to the STL/STEP filename

d-portero, 2023-08-18:
Every other slicer I have used (Simplify3d, Cura, Prusa/Superslicer) defaults to exporting the print files with the filename of the original STL. BambuStudio defaults to "Untitled". If the filename is not changed from "Untitled" the previous print job is overridden. 

Please implement that the slicer defaults to the STL filename if there is no project name already.

bambu123, 2023-08-21:
This problem is relate to send/print to the printer.
Save 'Untiled' project will use the name of the  last added model.

Use plate name is another good suggestion.

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl