bambulab/BambuStudio #2327: Bug Report: 1545 has been closed in Error - The problem still exists.

rdwrigh2, 2023-08-18:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU Intel i9
64 meg of RAM

### Printer

Bambu Lab X1C

### How to reproduce

Just look back to Bug Report: 1545.  The problem is exactly the same as when it was reported back in March.  In short, if you add text to an object, you can edit it as long as you HAVE NOT saved and reopened the project.  Once you save and reopen a previously saved project and then try to "Edit" a previously added text, you CANNOT because the "Edit Text" right-mouse-click menu choice is "disabled" (greyed out.  Whoever closed the ticket stated that the problem has been solved in version 1.60.  I has not been fixed in version 1.60  (I just downloaded it to make sure) nor has it been fixed in any version since then, including the latest version 1.7.4.52.

I really appreciate all of the new features that keep getting added to Bambu Studio, but this is a very annoying problem because you have to DELETE any previously added "texts" and recreate them from scratch which is very time consuming.

### Actual results

Refer to Bug Report 1545 for examples.  The problem has NOT been solved.

### Expected results

I would expect the right-click "Edit Text" menu choice NOT to be disabled when you open a project that has already had "text" added to an object.  I want the "Edit Text" to be "Enabled" and allow the user to change (Edit) any text that has been previously added to an object before the project as saved.

Note: You can edit any text you have added to an object "to your hearts content" as long as you HAVE NOT saved and closed the project.  Once the project has been saved and reopened, you then can NO LONGER use the "Edit Text" menu/feature.

### Project file & Debug log uploads

[PushStickLrgeHandle - Flat with Name.zip](https://github.com/bambulab/BambuStudio/files/12375366/PushStickLrgeHandle.-.Flat.with.Name.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-08-18:
@rdwrigh2 
If the text is generated by an old version of the software, there is no text information stored in the model, and the text cannot be edited even if it is opened with a new version.
You can use the new version of the software, add text and save it, so that you can edit the text when you open it with the new version

zhimin-zeng-bambulab, 2023-08-25:
@rdwrigh2 Is the text is generated by an old version?

zhimin-zeng-bambulab, 2023-08-30:
@rdwrigh2  Could you check if this problem still exists.

zhimin-zeng-bambulab, 2023-09-05:
[image]
After opening the file, select the text part, then right-click and select "Edit Text".

closed 2023-09-05 by zhimin-zeng-bambulab