bambulab/BambuStudio #2329: [Feature Request] - Lower voltage in G-Code for the printer

SnickCH, 2023-08-18:
**Is your feature request related to a problem? Please describe.**
On the printer specific settings in the slicer, there is a voltage specified by default. This seams very high. The motors are getting exremly hot in the chamber. 

I saw that many people changed the voltage down by 10 - 20% to get them a bit coller. I also did some test print with -10% and had no issue. This high voltage makes probably only sense if you print in the ridiculous mods for the torque. But is this really needed on the normal mods? 

**Describe the solution you'd like**
I would kindly ask bambulab to investigate this to reduce the heat from the motors, by defining a lower voltage to the motors. This could maybe be a second printer profile for "standard" speeds.

**Describe alternatives you've considered**
Everybody can do this manually in the printer setup. There is nothing that holds us back. But I would like an official version or at least statement (like in the bambu wiki) that this was tested by bambulab. 

**Additional context**
The chamber already gets really hot even for PLA and PETG. Therefore everything that reduces the heat will help the electronics to life longer.

lanewei120, 2023-08-21:
@SnickCH 
hello, thanks for the suggestion

The current voltage/current configuration is the most suitable value we have adapted for different environmental conditions, materials, models, and printing speeds. If it's reduced, we cannot guarantee universality anymore.

SnickCH, 2023-08-21:
@lanewei120  Thank you for your feedback

You could take a look on it for the "standard printign speed". I think this voltage was optimized for the high printingspeeds - only there you need this much torque. 

Netherless I will close this request, because it looks like it will not be looked further.