bambulab/BambuStudio #2334: Front Nozzle Load Line is not using first layer temperature

brandonraineri, 2023-08-18:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Additional system information

N/A

### Printer

Bambu X1C

### How to reproduce

Change **First Layer Temperature** to something lower than **Other Layer Temperature**.

### Actual results

When printing the Front Nozzle Load Line, it is using **Other Layer Temperature** instead of **First Layer Temperature**.  The back right purge line is using the correct **First Layer Temperature**.  I have not confirmed if the calibration lines use the higher temperature or not.

### Expected results

Everything printing on the first layer should use **First Layer Temperature**.  I print much hotter on other layers to get good layer adhesion.  When you print anything on the first layer in these temperatures, the bond to the plate and prevent you from easily removing them, sometimes, resulting in damaging the plate.

### Project file & Debug log uploads

[FrontNozzleLoadTemperatureBug.zip](https://github.com/bambulab/BambuStudio/files/12383745/FrontNozzleLoadTemperatureBug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zviratko, 2023-08-20:
I believe that first line is for calibration purposes and and then it makes sense to not make it at the first layer temperature as the properties (like in case with LW filaments) might be different. I noticed this also but I believe it to be intentional and beneficial. I doubt layer adhesion matters much for this line.

I also have filaments that bond too tighty and use a gluestick as a separation layer, I suggest you do the same.

QingZhangBambu, 2023-08-25:
Calibration is for normal printing, so the nozzle temp should as same as the normal print temp.

SaltWei, 2024-01-16:
> Everything printing on the first layer should use **First Layer Temperature**.

No. Temperature has influence on PA. Should use the most widely used temperature(Other layer) to do pa calibration.