bambulab/BambuStudio #2335: Bambu studio keeps freezing even though the computer is up to date and the bambu studio is up to date please help

joeymicallef, 2023-08-18:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac ventura 2.0

### Additional system information

apple m1 pro cpu
16 gb ram

### Printer

X1 carbon

### How to reproduce

1. turn on computer
2. double click on Bambu studio (the most recent update) to open the application
3. import 1st model  into Bambu studio and all is good i am able to edit and move the item or model around
4. when the 2nd file gets imported the entire Bambu studio freezes I can not close the application either (both Bambu studio and the Macbook are up to date)
5. I go to slice and it wont slice either and i can not click on anything on the Bambu studio
6.The only way to get the Bambu studio application to work again is to shutdown and restart the computer is needed to restart the application because the application is frozen and i can not click the x in the top left to close the app to restart the app
7. it almost seems like there was not enough time spent on the development of the application for apple mac computers

### Actual results

i would take a screen shop but your Bambu studio application freezes when a file is uploaded

### Expected results

the bambu labs x1 carbon use to print flawless with multiple parts, but i cant get the Bambu studio to unfreeze when i upload a file to the Bambu studio

### Project file & Debug log uploads

cant get the files due to the application being frozen and not being able to un freeze the application $HOME/Library/Application Support/BambuStudio/log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-08-19:
@joeymicallef 

could you give a short video for your operations?
also please help to upload the logs from  /Users/**user.name**/Library/Application Support/BambuStudio/log
you can zip this directory and upload here

lanewei120, 2023-09-06:
@joeymicallef do you still have this issue?
if yes, could you give more information just as I mentioned above

utiko, 2023-09-22:
I have the same issue. It is freezing on launch on M1:
[image]

Here are my logs.
[log_latest.zip](https://github.com/bambulab/BambuStudio/files/12703732/log_latest.zip)

utiko, 2023-09-22:
However, I found a workaround to launch it.
1. Disable Wifi
2. Launch the app
3. Enable Wifi  
:)

ackerman76, 2023-10-21:
If I send another file to the printer with the same name, then go to print it from studio, and I cancel the current print job, Bambu slicer will freeze with the Print popup "Fetching model information", even after I press cancel once, then it locks and it cannot recover at all, even if the printer is fine and I can interact with it. I have to ctrl alt del and kill studio, It will not exit. It has never done this before.

Latest X1-C firmware 01.06.05.01 and Latest bambu slicer / studio as of today 1.7.7.89.

[image]

lanewei120, 2023-10-21:
> If I cancel the current print job, send another file to the printer with the same name, then go to print it from studio, Bambu slicer will freeze with the Print popup "Fetching model information", even after I press cancel once, then it locks and it cannot recover at all, even if the printer is fine and I can interact with it. I have to ctrl alt del and kill studio, It will not exit. It has never done this before.
> 
> Latest X1-C firmware 01.06.05.01 and Latest bambu slicer / studio as of today 1.7.7.89.
> 
> [image]

@ackerman76 
could you help to upload the logs here?
just following this guide
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

bambu123, 2023-10-21:
This is a known issue. When coming back to model list, please wait a while before starting   print.

bambu123, 2024-02-06:
> However, I found a workaround to launch it.
> 
> 1. Disable Wifi
> 2. Launch the app
> 3. Enable Wifi
>    :)

Do you still have this problem on the latest version of Studio? If so, please refer to the wiki below to provide us with a Mac report.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue