bambulab/BambuStudio #2336: By Object with color change has first layer under extrusion

roflkins, 2023-08-19:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 22H2 22621.2134

### Additional system information

CPU: Intel i9-12900k
RAM: 64 GB DDR5-6400
GPU: Nvidia RTX 3070 Ti

### Printer

BambuLab X1C (01.06.01.00) with 1 AMS

### How to reproduce

Have 2 different color filaments loaded into the AMS. If not already done, run Calibration -> Flow Dynamics -> Auto-Calibration. Ensure global -> others -> print sequence is set to by object. Slice plate and print with bed leveling enabled.

### Actual results

Resulting second object printed has a visible gap in the first layer. The first object printed, which uses the filament from the prime lines bordering the plate, does not have the defect.

[image]

### Expected results

Both objects should have no under extrusion issues or gaps in the first layer. Note this bug does not show up if by object is used when the filament for both cubes are the same. 

[image]

### Project file & Debug log uploads

[Bottom Extrusion Test.zip](https://github.com/bambulab/BambuStudio/files/12384673/Bottom.Extrusion.Test.zip)
[debug_network_Wed_Aug_16_16_12_00.log](https://github.com/bambulab/BambuStudio/files/12384674/debug_network_Wed_Aug_16_16_12_00.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

roflkins, 2023-08-19:
A hunch. This occurs due to lack of priming when going from color to color, since using a single color doesn't yield the defect. Also worth noting that this defect show up in the same place regardless of print plate (high temp or cool plate) used.

zhimin-zeng-bambulab, 2023-08-25:
> A hunch. This occurs due to lack of priming when going from color to color, since using a single color doesn't yield the defect. Also worth noting that this defect show up in the same place regardless of print plate (high temp or cool plate) used.

@roflkins 
Yes, you are right, because there is no prime tower for by-object printing, when the color is changed, there is a shortage of material at the beginning of printing, and the location of the defect is related to the order of inner wall/outer wall/infill.

SaltWei, 2024-01-18:
This is bug of gcode export. One more retraction is done when finish one object and move to next