bambulab/BambuStudio #2339: Filament settings got lost when loading a *.3mf file

Panda631, 2023-08-19:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

X1C & 2xAMS

### How to reproduce

1. Resync all filements from AMS in the prepare tab. In my case, I have 2 AMS attached to the X1C
[image]

2. Open the attached test file. 
[TEST BambuLab X1 - Temperature tower (PETG 250-290).zip]

3. All current filaments are replaced by totaly different filaments. 

4. Only one AMS is shown in the overview.

[image]

### Actual results

All filaments are replaced by other types. I supect it came with the 3mf file.
Only one AMS is displayed although two are connected.

### Expected results

Current filaments should NOT be replaced by other filament types.
All attached AMS devices should be visible.

### Project file & Debug log uploads

[TEST BambuLab X1 - Temperature tower (PETG 250-290).zip](https://github.com/bambulab/BambuStudio/files/12385711/TEST.BambuLab.X1.-.Temperature.tower.PETG.250-290.zip)

[debug_network_Sat_Aug_19_10_34_16.zip](https://github.com/bambulab/BambuStudio/files/12385714/debug_network_Sat_Aug_19_10_34_16.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-08-21:
@Panda631 

hello, after loading a 3mf
all the filaments settings will be covered by the settings used in the 3mf
this can make sure all the settings in the 3mf are the same when it is shared to different persons

if you want to set to all the filaments on the ams you can 'sync ams' again

Panda631, 2023-10-26:
Please let me push this issue again.
Especially since you can import models direct from the Maker World to BambuStudio it quite annoying.

Yes, you can sync filaments from the AMS back to BambuStudio. But this doesn't work with filament brands others than Bambu. I use mainly SunLu filamants with specific settings and calibrations. After syncing with the AMS all filamant types are replaced with generic PLA. So I must change all filamant entries manuel back to my own SunLu type. This is quite annoying.

The same is for printer type and plate type.

My proposal is to ask the user whether he wants to use the settings in the 3mf file or keep its own settings.

MIKOLAJBF, 2024-04-06:
Panda631 is correct, this is quite irritating. Utilizing "sync ams" will synchronize with AMS filament settings, which are significantly limited compared to the settings in Bambu Studio. It should be chosen by the user in the preferences whether they desire automatic filament swaps when loading file or not.

gitterman, 2024-11-30:
I want to stress on this irritating issue even more.
The thing is, that a loaded .3mf file does not only change the filaments for the actual project, but (even worse) set this as default for all new projects.
This happens even if I close the project without accepting any modifications from the project.
This should urgently be fixed.

das303, 2024-12-05:
I'd like to give this a push, too. Same issue without AMS.
Bambu Studio 1.10.1.50
MacOs 14.6.1 (23G93)

When starting Bambo Studio I have a couple user defined Filaments available:
[image]

When opening e.g. https://makerworld.com/en/models/838758#profileId-784496 in BambuStudio, most of the most recent added custom Fillaments are gone, but some are still there.
[image]

mikypilota, 2025-01-19:
+1 for this issue, it is the default behaviour when importing directly from MakerWorld, and it should not be, as filament settings are going to differ from user to user.

davidarroyo1234, 2025-03-02:
Yep, this is annoying... And if you import the STL doesn't register this print as printed in makerworld etc etc...