bambulab/BambuStudio #2340: Seam generation: add "Sharpest" corner option

smokeandlights, 2023-08-19:
It would be really nice if the z seam generation could follow the sharpest corner of an object. This is a great way to hide seams, and it is laborious and messy using the seam painting tools.

QingZhangBambu, 2023-08-21:
i will add it to our feature pool and concerned staff will evaluate it

NVNDO, 2024-03-30:
I totally agree. Many objects would hugely benefit of this and it's a basic feature in all slicers. 

Existing Seam-options are not ideal. See here:

[image]

[image]

Being able to paint a seam is nice (but can still look awkward).
See here: 

[image]

[image]

Seam painting should not be the go to option, rather an alternative if nothing else works.

Pls implement.

NVNDO, 2024-04-02:
Sitting here since half an hour trying my best, but nothing. 

[image]

Even using the seam-tool, the results are very bad! 

[image]

@QingZhangBambu Please implement. 
It's a standard feature everywhere else.

NVNDO, 2024-04-09:
I'm done with this!! 

I changed to Orca Slicer. 
There at least back is = back. as not even that functions works properly in Bambu Studio...

[image]