bambulab/BambuStudio #2342: "Ensure vertical shell thickness" setting is not functional in some situations

SupMap, 2023-08-19:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma

### Printer

Bambu Lab x1c + AMS

### How to reproduce

Open Bambu Studio and import the attached model. Slice with the default 0.08mm profile for PLA (any), with the options "Ensure vertical shell thickness" unchecked and "Minimum sparse infill threshold" set to "0". 

### Actual results

You will see that solid infill is still being generated on the inner side of the walls of the model, in areas with a higher angles more infill is generated. This can be reduced only by lowering "Bottom shell layers and thickness" to 0, but then you have no bottom shell. 

### Expected results

If the option "Ensure vertical shell thickness" is unchecked, no solid infill is generated around wall surfaces.

### Project file & Debug log uploads

[curve-overhang-test20180415-3629-5lpeqk.zip](https://github.com/bambulab/BambuStudio/files/12386326/curve-overhang-test20180415-3629-5lpeqk.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SupMap, 2023-08-20:
Additional comment:

As from my further investigation and the bug report "#1586" including the reply by "SaltWei", i was able to figure out what exactly causes the solid infill patterns to be added. However from my point of view and understanding, the option "Ensure vertical shell thickness" should find application exactly in this case, by completely eliminating solid infill on walls (shells). 

Solid infill often causes more problems than it solves, may it be for extra light models, or due to longer print time and more filament consumption. Tweaking the settings top and bottom shell layers for each model until you end up without solid infill is time consuming and tricky, thus i see it as a huge improvement if the option "Ensure vertical shell thickness" would eliminate solid infill on walls/shells completely. Alternatively it would be possible to add another option, say "No solid infill on shells" which would do just that. 

As i see from the amount of similar requests/bug reports and Reddit posts, this is indeed an often requested change so i would appreciate a constructive reply on this!

BambooBugFixer, 2023-08-28:
The option "Ensure vertical shell thickness" is usually used for sloping surfaces to ensure the vertical shell thickness by having more solid infill. When this option is disabled, solid infill is generated according to top shell layers, top shell thickness, bottom shell layers and bottom shell thickness as SaltWei has explained. You can reduce solid infill to the largest extend by setting shell thickness to 0 (disabling this setting) and setting top and bottom layers to a relatively low value instead of zero, while still retaining top and bottom shells. We will take "No solid infill on shells" option seriously and decide if we'll have such option in Bambu Studio.

SupMap, 2023-08-28:
Thank you very much for your feedback, i absolutely understand the theory behind why this solid infill is added. Also i appreciate that you will consider adding the option "No solid infill on shells" to the slicer - while searching for a solution i have found quite many users who were searching for the same solution, so i suppose this will be a frequently required option and it should be fairly easy to implement.

closed 2025-05-30 by QingZhangBambu