bambulab/BambuStudio #2343: Project Tab broken

RaySajuuk, 2023-08-19:
### Bambu Studio Version

1.7.4

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13

### Printer

Bambu Lab X1CC

### How to reproduce

Open existing 3mf file/Import model file in new project.
Go to project tab.

### Actual results

Unable to edit any information

### Expected results

Back in 1.5.x I was able to edit project information.

### Project file & Debug log uploads

[Project Info Test.3mf.zip](https://github.com/bambulab/BambuStudio/files/12386666/Project.Info.Test.3mf.zip)
[debug_Fri_Aug_18_21_35_06_5082.log.0.zip](https://github.com/bambulab/BambuStudio/files/12386667/debug_Fri_Aug_18_21_35_06_5082.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MadMax389, 2023-08-19:
Please bring back the project tab. I miss it!

lanewei120, 2023-08-21:
we are re-design this logic, and hope it will come out soon

RaySajuuk, 2023-08-21:
> we are re-design this logic, and hope it will come out soon

If the function is not working why don't temporarily remove it or at least add some type of notice on the tab or wiki. The lack of proper communication is frustrating, the time I spent on trying to figure out what is wrong on my end was completely wasteful. Please be respectful to all the users in the community that actively trying to support you guys. Thanks.

cpasoft, 2023-09-25:
One month later, no news and the tab still not working...

YifanWuBambu, 2023-10-10:
Hi @RaySajuuk 
The project tab in bambu studio is now read only.
It displays information of the models downloaded from MakerWorld.
[image]
[image]
[image]

If you want to insert informations into your 3mf, please upload the 3mf as a model in makerworld.com.
When it is published, you may download the 3mf from the model page and all informations will be added to the 3mf.

If you do not want others see the 3mf, you may set it as private visibility.
[image]

RaySajuuk, 2023-10-13:
> Hi @RaySajuuk The project tab in bambu studio is now read only. It displays information of the models downloaded from MakerWorld. [image] [image] [image]
>
> If you want to insert informations into [...]

@YifanWuBambu This is nice to change to see. I do the fully functioned project information editor comes back soon. It is nice to have the integration with MakerWorld, which I assumed it will be the case. It is also a very useful tool beyond the current read-only for MakerWorld models.

lanewei120, 2023-10-24:
close this issue as the solution is published

closed 2023-10-24 by lanewei120

MadMax389, 2023-10-24:
I hope this is not the permanent "fix" for this function. I still cannot edit or add any information in the project tab. I lack the ability to add or change the order of any picture, set the cover photo or add text or pdf files. The aspect ratio of the cover picture is way off. To me, it is impractical to upload the profile to MW and then download it. I always modify my uploaded profiles with standard printer/filament profiles to pass the upload screening test, but prefer to keep my local copies with my own modified profiles. I always delete the downloaded copies.

RegexRiddler, 2023-10-27:
@lanewei120 How can you call that a solution? The issue is that we can't write that information directly in the project tab in Bambu Studio, forcing the users to upload the files to MakerWorld and then download them again isn't a solution at all.

As a product developer I cannot upload any of my files to any services no matter how "private" they are. And even for personal projects, if I make changes to the models I then have to re-upload the new file and then download it again.