bambulab/BambuStudio #2346: Incorrect button text when loading filament

gsingh93, 2023-08-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.5

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to the Device tab
2. Click Load

### Actual results

The dialog text refers to the "Done" and "Retry" buttons (which appear on the printer display), but the dialog buttons are "Done" and "Confirm":
[image]

### Expected results

The buttons should be "Done" and "Retry", and "Done" should probably be the green one. I wasn't paying attention and just clicked the green button, assuming that would continue the loading process, and I was confused when things didn't work. After trying it a second time I realized I needed to click the other button.

### Project file & Debug log uploads

Not needed for UI issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-08-23:
Hi.
We will optimize the highlighting of the buttons

walterwongbbl, 2023-08-24:
@gsingh93 
Highlighting the "Done" button can reduce misunderstandings. We will optimize it.

walterwongbbl, 2023-10-13:
@gsingh93 
Please download and use the latest Studio version

closed 2023-10-13 by walterwongbbl