bambulab/BambuStudio #2351: Resume button missing after inserting pause in slicer at specific layer

timtimellis, 2023-08-21:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.4.1 (c)

### Additional system information

Apple Silicon, Macbook Pro M1 2021, 32GB RAM

### Printer

Bambu Lab X1CC

### How to reproduce

I have not tested being able to reproduce this repeatedly yet however:

1. Insert a pause at a specific layer, using the slider in the Preview part of Bambu Studio
2. Run print
3. Print stops, it tells me on various interfaces (Bambu Handy, on the printer, in Bambu Studio) that I need to tap resume to continue

### Actual results

There is no resume button available in Bambu Studio, the mobile app or the printer, and none of the workarounds suggested on reddit or elsewhere work in this case. I was able to make the resume button reappear in Bambu Handy by clicking pause in the main window, then pause in the modal, but on hitting resume ......... it did not work.

[image]
[image]
[image]
[image]

### Expected results

There should be some way to resume the print. There is not. It's in the bin, like the overall quality of my Bambu Labs experience recently.

### Project file & Debug log uploads

The file contains customer intellectual property and will not be provided.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-08-22:
Very stupid bug. But not caused by APP and studio, but firmware.
@sodachen22 Please check this issue with high priority. User can not resume the task after pause.

sodachen22, 2023-08-22:
There is a release of 01.06.03.00, would you please have a test on the latest version?

FyrbyAdditive, 2023-08-22:
Thank you, this firmware seems to not be available in Bambu Studio for some reason (another bug, or perhaps slow CDN propagation, no idea), however I will return to the printer soon, maybe the firmware update will be there and I will try and recreate it hopefully soon.

FyrbyAdditive, 2023-08-22:
All updated (via printer interface) - test is running and will report back soon.

FyrbyAdditive, 2023-08-22:
I am unable to recreate this using the same project and method on 01.06.03.00. Thank you for the quick attention.

closed 2023-08-22 by FyrbyAdditive

roblomq, 2024-03-24:
A few months later and it seems the bug is still not fixed, screwed up two prints because I couldn't resume the print.
Bug is on a X1C with the latest firmware.
[image]
[image]