bambulab/BambuStudio #2352: printable checkmark in object list is broken

kirillfx, 2023-08-21:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.3.1 (a) 

### Printer

Bambu Lab P1S

### How to reproduce

1. Import multiple objects
2. Start to click 'printable' checkboxes in different order

### Actual results

1. For some objects 'printable' does opposite.
2. Affects wrong objects.

### Expected results

Expected working 'printable' checkboxes.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12394472/log.zip)

I can't include original project file since undo function for 'delete plate' operation is broken too.

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-08-25:
@kirillfx Please note that the first click is to select the part and does not change the state of the checkbox.

bambu123, 2023-10-31:
Closed because the user did not reply for a long time.

closed 2023-10-31 by bambu123