bambulab/BambuStudio #2355: [Request] Add Troodon 2.0 to the supported printers list (already in OrcaSlicer)

nirurin, 2023-08-21:
**Is your feature request related to a problem? Please describe.**
No default printer option for Troodon 2.0. 

**Describe the solution you'd like**
Copy it in from Orcaslicer 

**Describe alternatives you've considered**
Using orcaslicer instead at the moment

SaltWei, 2023-08-22:
@XunZhangBambu Please add it in next version

bolsoncerrado, 2023-10-05:
I'd like to see Troodon v1.0 added as well.

nirurin, 2023-12-22:
> @XunZhangBambu Please add it in next version

@SaltWei @XunZhangBambu 

I'm noticing that several/many versions have been released, and the Troodon2 still hasn't been transferred over from orca. Is there some issue, or was it just forgotten about? lol

XunZhangBambu, 2023-12-25:
In next version ,you will see Troodon 2.0

XunZhangBambu, 2024-01-22:
In latest version, we have added Troodon 2.0

closed 2024-01-22 by SaltWei