bambulab/BambuStudio #2356: Update manual 'flow calibration' to the modern method

nirurin, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
The manual 'flow calibration' method on bambu studio is an old and really badly designed method. 

**Describe the solution you'd like**
Update to the newer 'Ellis' method, which Orcaslicer now uses - https://github.com/SoftFever/OrcaSlicer/pull/1547

[image]

SaltWei, 2023-08-22:
Already add it for 3-rd party printer in 1.7.4.
@zhimin-zeng-bambulab Please add it for manual calibration in next version

SaltWei, 2023-08-22:
[image]
@nirurin In 1.7.4， you can choose 3-rd party printer first, and enable the pa pattern. And then switch to Bambu printer to use it.

We will add the pattern in Calibration page directly in next version.

MartynWilliams, 2023-09-22:
@zhimin-zeng-bambulab this appears not to be live in version 1.7.6.92, can you please confirm this omission.

closed 2023-09-25 by zhimin-zeng-bambulab