bambulab/BambuStudio #2358: Filament setting override to control chamber fan

Belowthoc, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

 when I print petg I find that cooling and chamber temp are highly important. currently, I cant find an easy way to set chamber fan speed unless altering filament g-code or changing it once the print has started.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A simple button to activate and deactivate the chamber fan in the filament setting. When activated, you can input a percentage amount of how fast you want it to run.

SaltWei, 2023-08-22:
> chamber fan

Which fan？ The fan in the back of chamber？

Belowthoc, 2023-08-22:
Yes. The one that exhausts the chanber.

XunZhangBambu, 2023-08-23:
Next version we will add the feature

bani6809, 2023-09-07:
Please add this feature.

I have an external BOFA fume extractor attached to my X1C so I don't need chamber fan interfering with it, I'd like to turn chamber fan OFF.

Belowthoc, 2023-09-07:
Glad you think its a good idea also. It would he such a simple addition.
And why not give controll over every aspect of the print process. Vent fan
is part of that process.

XunZhangBambu, 2023-10-10:
[image]

In the latest version V1.7.7, you can control chamber fan by controlling exhaust fan in Bambu printers . Using exhaust fan to enhance air filtration is a new feature of X1E  and haven't been thoroughly tested in other printers , and currently "support air filtration" is enabled by default only for X1E. So you need to open develop mode and enable "support air filtration" if you want to control exhaust fan in other Bambu printers.

closed 2023-11-14 by QingZhangBambu