bambulab/BambuStudio #2360: No context in alert window.

werewolfenn, 2023-08-22:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 pro, 64 bit

### Additional system information

Shouldn't matter for this issue.

### Printer

X1C

### How to reproduce

Be in an open project.    Click to start a NEW project.   This opens up asking if you want to discard or transfer changes made.

### Actual results

Box asking about discarding or transferring changes doesn't tell you what those changes are.   Relatively useless message.

### Expected results

If we are being asked if we wish to save or discard setting changes, we need to know where those changes apply.
[image]

### Project file & Debug log uploads

[debug_network_Thu_Aug_17_19_14_41.log](https://github.com/bambulab/BambuStudio/files/12410124/debug_network_Thu_Aug_17_19_14_41.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-08-24:
@werewolfenn could you shared your project 3mf file with me?

bambu123, 2023-08-24:
> @werewolfenn could you shared your the failure project 3mf file with me?

bambu123, 2023-12-21:
Fixed in v1.8.2.

closed 2023-12-21 by bambu123