bambulab/BambuStudio #2362: Supports > Tree > Style Default - Is Regular Support, Not a Tree

jason-e-gross, 2023-08-22:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

12th gen I-7
Geforce 3060
64GB RAM
2TB SD

### Printer

p1p

### How to reproduce

Go to Support
Enable Tree (auto)
Default Style

-- its a normal support, not a tree at all.

### Actual results

It's a normal support, not a tree

### Expected results

A tree

### Project file & Debug log uploads

[project-and-screenshots.zip](https://github.com/bambulab/BambuStudio/files/12413650/project-and-screenshots.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

jason-e-gross, 2023-08-22:
[image]
[image]

jason-e-gross, 2023-08-23:
[image]
[image]

NAM1025, 2023-08-23:
This is happening for me as well and I'm on 1.6.2.4. Support critical regions only gets rid of it, but I need the critical regions to be supported as well so it doesn't solve the issue for me.

[image]

tangketan, 2023-08-24:
Default style of tree support is "tree hybrid", which is the combination of tree and normal supports. When an overhang area is large and flat, normal supports will be generated.
If you don't like the style, just change it to tree slim or tree strong.

closed 2023-08-24 by tangketan

namarinelli, 2023-08-24:
> Default style of tree support is "tree hybrid", which is the combination of tree and normal supports. When an overhang area is large and flat, normal supports will be generated. If you don't like the style, just change it to tree slim or tree strong.

Oh, I didn't realize. I assumed "Tree Hybrid" was a combination of "Tree Slim" and "Tree Strong"

jason-e-gross, 2023-08-24:
Same -- having "Default" _and_ "Tree Hybrid" in there makes it confusing.   If you choose "Tree (auto)" maybe get rid of "Default" and have it simply default to "Tree Hybrid" with a tool tip that explains the app will choose between normal and tree.

Because now you're _explicitly_ picking "Tree (auto)" and don't expect to see non-tree generation inside it.

I wouldn't agree that this is "Completed", because the UI is misleading and what it says for **Default** is different than what it describes for "Tree Hybrid" (a mix of normal and tree).