bambulab/BambuStudio #2363: Color Painting > Fill: Ignores STL dimensions

cutmoney, 2023-08-23:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 13.4.1

### Additional system information

MacBook Pro M1 Max 2021

### Printer

Bambu Lab P1P

### How to reproduce

1. Import model with multiple bodies intended for multi-color.
2. Utilize the Color Painting > Fill tool to colorize specific parts of the model.

### Actual results

As is indicated in the attached project file and images, the model has the AB letters that extend 5mm in to the body. Bambu Studio only extends the color .6mm deep, and the outline 1.0mm deep. The actual model has the letters cut 5mm deep. Is this a bug? Is there any way for Bambu Studio to recognize the actual size of the bodies that I want colorized? Or to specify the depth of the color?

[image]
[image]
[image]
[image]
[image]

### Expected results

The color would extend to the entire body depth of the selected feature. In this case, 5mm deep instead of the .6mm deep. Alternatively, there is an option to select how deep the color would go.

### Project file & Debug log uploads

[AB Test.3mf.zip](https://github.com/bambulab/BambuStudio/files/12414137/AB.Test.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-08-25:
@cutmoney  The 3mf file you made only has coloring information, and the 3mf file does not contain the size information of the infill part.

cutmoney, 2023-08-28:
> @cutmoney The 3mf file you made only has coloring information, and the 3mf file does not contain the size information of the infill part.

That was the Bambu Studio file, and it works identically with any object. Nevertheless, here is an example 3mf...
[test.3mf.zip](https://github.com/bambulab/BambuStudio/files/12458798/test.3mf.zip)

zhimin-zeng-bambulab, 2023-08-29:
[image]

@cutmoney  In this case, you do not need to paint, you can  set  Body2(A) and Body3(B) to white color. 
As shown in FIG.

zhimin-zeng-bambulab, 2023-09-01:
@cutmoney Do the above steps solve your problem?

zhimin-zeng-bambulab, 2023-09-05:
If you want the sub-part to display the corresponding color, you should set the default color of the sub-part instead of painting the surface.

closed 2023-09-05 by zhimin-zeng-bambulab

cutmoney, 2023-09-05:
> If you want the sub-part to display the corresponding color, you should set the default color of the sub-part instead of painting the surface.

Being able to colorize individual parts is exactly what I wanted. Thanks. When I export directly from Fusion 360 it comes in as a single body, however when I export and import as a file, it allows for the different parts. Thanks for the info, that's helpful!